luxas/kubernetes-on-arm #100: Support for Ingress Resource

larmog, 2016-05-14:
Is there any plans for, or is there currently any work going on, for adding support for Ingress Resources, instead of the `loadbalancer` addon?

luxas, 2016-05-14:
It would be really cool with the ingress resource!
I checked it up, and we should probably use this controller: https://github.com/kubernetes/contrib/tree/master/ingress/controllers/nginx
I've never experimented with ingress, but soon I'll check it out

In the first place, I'll send a PR to kubernetes/contrib that cross-compiles all their ingress images to arm, arm64 and ppc64le, instead of implementing it in this repo

larmog, 2016-05-14:
Let me know if there's anything I can do to help.
/Lars