luxas/kubernetes-on-arm #113: Include nfs-common in hypercube

mitchese, 2016-07-20:
I am trying to use NFS for a persistent volume in a container. I am getting the error "wrong fstype" when trying to deploy the container.

I've installed nfs-common in both the container itself, as well as the base OS with no luck. After hopping into the hypercube image via docker exec exec and installing it, it seems to be working.

Mounting the nfs share on the host works fine, so it's not that it's missing on the host.

Relevant section of the error log: 
```
Events:
  FirstSeen	LastSeen	Count	From				SubobjectPath	Type		Reason		Message
  ---------	--------	-----	----				-------------	--------	------		-------
  15s		15s		1	{default-scheduler }				Normal		Scheduled	Successfully assigned gitlab-o5iay to 192.168.141.42
  15s		1s		2	{kubelet 192.168.141.42}			Warning		FailedMount	Unable to mount volumes for pod "gitlab-o5iay_default(c052ee18-4ea6-11e6-8652-b827ebb55e21)": Mount failed: exit status 32
Mounting arguments: 192.168.141.41:/media/docker/gitlab /var/lib/kubelet/pods/c052ee18-4ea6-11e6-8652-b827ebb55e21/volumes/kubernetes.io~nfs/data nfs []
Output: mount: wrong fs type, bad option, bad superblock on 192.168.141.41:/media/docker/gitlab,
       missing codepage or helper program, or other error
       (for several filesystems (e.g. nfs, cifs) you might
       need a /sbin/mount.<type> helper program)

       In some cases useful info is found in syslog - try
       dmesg | tail or so.

  15s	1s	2	{kubelet 192.168.141.42}		Warning	FailedSync	Error syncing pod, skipping: Mount failed: exit status 32
Mounting arguments: 192.168.141.41:/media/docker/gitlab /var/lib/kubelet/pods/c052ee18-4ea6-11e6-8652-b827ebb55e21/volumes/kubernetes.io~nfs/data nfs []
``` 

From the Dockerfile of the container: 
```
      volumes:
      - name: data
        nfs:
            server: 192.168.141.41
            path: /media/docker/gitlab
```

Would this be an acceptable solution for mounting persistent volumes, or is there a better way that I'm missing?

luxas, 2016-07-22:
Thanks for the issue.
I'm gonna use the official hyperkube images now in the next release, so I suggest you should make a PR against official Kubernetes and add `nfs-common` there.

saturnism, 2016-08-26:
In the mean time, I had to re-append `--containerized` option for kubelet starts, with docker volume mount `-v /:/rootfs:ro`

luxas, 2016-09-10:
nfs-common will be available in newer versions. In the meantime, upgrade to v0.8.0 and add `export USE_CONTAINERIZED=true` to `/etc/kubernetes/k8s.conf` and install `nfs-common` right on host

closed 2016-09-10 by luxas

saturnism, 2016-09-11:
thanks @luxas!