luxas/kubernetes-on-arm #115: Not finding master node

theHound9987, 2016-07-27:
I'm trying to connect worker node to my master and am getting the following output:
kube-config enable-worker 192.168.1.235
Disabling k8s if it is running
Using master ip: 192.168.1.235
The Kubernetes master was not found.
Exiting...

Command:
kube-config enable-worker [master-ip]

I can ping my master any ideas?

root@cluster-slave-node-01:/home/pirate# ping 192.168.1.235
PING 192.168.1.235 (192.168.1.235) 56(84) bytes of data.
64 bytes from 192.168.1.235: icmp_seq=1 ttl=64 time=0.544 ms
64 bytes from 192.168.1.235: icmp_seq=2 ttl=64 time=0.421 ms
64 bytes from 192.168.1.235: icmp_seq=3 ttl=64 time=0.372 ms
64 bytes from 192.168.1.235: icmp_seq=4 ttl=64 time=0.441 ms
64 bytes from 192.168.1.235: icmp_seq=5 ttl=64 time=0.358 ms
64 bytes from 192.168.1.235: icmp_seq=6 ttl=64 time=0.441 ms
64 bytes from 192.168.1.235: icmp_seq=7 ttl=64 time=0.348 ms
64 bytes from 192.168.1.235: icmp_seq=8 ttl=64 time=0.561 ms
64 bytes from 192.168.1.235: icmp_seq=9 ttl=64 time=0.465 ms

jamesbuckett, 2016-08-01:
I had a problem when i first installed a worker. 

I found that I had to add this to the cmdline.txt to get the master to work.

sudo vi /boot/cmdline.txt
cgroup_enable=cpuset

After this change the master came up and then the workers were able to connect.

I made this change to all the pi's in my cluster.

luxas, 2016-09-10:
I've fixed this in `v0.8.0`

closed 2016-09-10 by luxas