luxas/kubernetes-on-arm #118: Working with Hypriot 1.0

saturnism, 2016-08-25:
A few things I found that needed to change when working w/ Hypriot 1.0:

1. Host name needs to be updated in `/boot/device-init.yaml`
1. docker-flannel overlay was not installed by default
1. `flannel.service` needs to be add `After=system-docker.service` to survive restarts
1. I still need to run kubelet with `--containerized=true` in order to use NFS
1. Need to append `/boot/cmdline.txt` with `cgroup_enable=cpuset`
1. Need to enable mtu probing to avoid docker pull problems: append `net.ipv4.tcp_mtu_probing=1` to `/etc/sysctl.conf` (optionally swappiness? see http://a.frtzlr.com/kubernetes-on-raspberry-pi-3-the-missing-troubleshooting-guide/)

pakeha-kiwi, 2016-08-31:
Thanks for this, am having issues running a Pi2/3 mixed cluster - OS OK - but Hypercube the only container running - issue connecting to the Apimanager

`$ docker logs 328725077e0f
W0831 02:50:35.960391    2113 server.go:445] Could not load kubeconfig file /var/lib/kubelet/kubeconfig: stat /var/lib/kubelet/kubeconfig: no such file or directory. Trying auth path instead.
W0831 02:50:35.960953    2113 server.go:406] Could not load kubernetes auth path /var/lib/kubelet/kubernetes_auth: stat /var/lib/kubelet/kubernetes_auth: no such file or directory. Continuing with defaults.
I0831 02:50:35.964233    2113 plugins.go:71] No cloud provider specified.
I0831 02:50:35.964534    2113 server.go:312] Successfully initialized cloud provider: "" from the config file: ""
I0831 02:50:35.966133    2113 manager.go:132] cAdvisor running in container: "/docker/328725077e0f6459c8c0d34bde92891b32309eafb8fa98360e822608d2a38e5e"
I0831 02:50:36.078324    2113 fs.go:109] Filesystem partitions: map[/dev/root:{mountpoint:/etc/resolv.conf major:179 minor:2 fsType: blockSize:0}]
E0831 02:50:36.104399    2113 machine.go:176] failed to get cache information for node 0: open /sys/devices/system/cpu/cpu0/cache: no such file or directory
I0831 02:50:38.858774    2113 manager.go:169] Machine: {NumCores:4 CpuFrequency:1200000 MemoryCapacity:903831552 MachineID:f21c916157f34d369cfa9e81943edba7 SystemUUID:f21c916157f34d369cfa9e81943edba7 BootID:e8f9bfac-a888-4451-a7b0-bb8008bfbbe1 Filesystems:[{Device:/dev/root Capacity:30100942848 Type:vfs Inodes:7120960}] DiskMap:map[179:0:{Name:mmcblk0 Major:179 Minor:0 Size:32010928128 Scheduler:deadline}] NetworkDevices:[{Name:eth0 MacAddress:b8:27:eb:57:92:40 Speed:100 Mtu:1500} {Name:wlan0 MacAddress:b8:27:eb:02:c7:15 Speed:0 Mtu:1500}] Topology:[{Id:0 Memory:0 Cores:[{Id:0 Threads:[0] Caches:[]} {Id:1 Threads:[1] Caches:[]} {Id:2 Threads:[2] Caches:[]} {Id:3 Threads:[3] Caches:[]}] Caches:[]}] CloudProvider:Unknown InstanceType:Unknown InstanceID:None}
I0831 02:50:38.864322    2113 manager.go:175] Version: {KernelVersion:4.4.15-hypriotos-v7+ ContainerOsVersion:Raspbian GNU/Linux 8 (jessie) DockerVersion:1.12.1 CadvisorVersion: CadvisorRevision:}
I0831 02:50:38.870499    2113 server.go:319] Using root directory: /var/lib/kubelet
I0831 02:50:38.872138    2113 server.go:673] Adding manifest file: /etc/kubernetes/manifests-multi
I0831 02:50:38.872477    2113 file.go:47] Watching path "/etc/kubernetes/manifests-multi"
I0831 02:50:38.882447    2113 server.go:683] Watching apiserver
W0831 02:50:38.910641    2113 kubelet.go:508] Hairpin mode set to "promiscuous-bridge" but configureCBR0 is false, falling back to "hairpin-veth"
I0831 02:50:38.911417    2113 kubelet.go:276] Hairpin mode set to "hairpin-veth"
I0831 02:50:38.994205    2113 manager.go:244] Setting dockerRoot to /var/lib/docker
I0831 02:50:38.994335    2113 plugins.go:56] Registering credential provider: .dockercfg
I0831 02:51:00.199644    2113 plugins.go:291] Loaded volume plugin "kubernetes.io/aws-ebs"
I0831 02:51:00.199911    2113 plugins.go:291] Loaded volume plugin "kubernetes.io/empty-dir"
I0831 02:51:00.200095    2113 plugins.go:291] Loaded volume plugin "kubernetes.io/gce-pd"
I0831 02:51:00.200276    2113 plugins.go:291] Loaded volume plugin "kubernetes.io/git-repo"
I0831 02:51:00.200462    2113 plugins.go:291] Loaded volume plugin "kubernetes.io/host-path"
I0831 02:51:00.201832    2113 plugins.go:291] Loaded volume plugin "kubernetes.io/nfs"
I0831 02:51:00.202104    2113 plugins.go:291] Loaded volume plugin "kubernetes.io/secret"
I0831 02:51:00.202423    2113 plugins.go:291] Loaded volume plugin "kubernetes.io/iscsi"
I0831 02:51:00.202888    2113 plugins.go:291] Loaded volume plugin "kubernetes.io/glusterfs"
I0831 02:51:00.203141    2113 plugins.go:291] Loaded volume plugin "kubernetes.io/persistent-claim"
I0831 02:51:00.203333    2113 plugins.go:291] Loaded volume plugin "kubernetes.io/rbd"
I0831 02:51:00.203521    2113 plugins.go:291] Loaded volume plugin "kubernetes.io/cinder"
I0831 02:51:00.203700    2113 plugins.go:291] Loaded volume plugin "kubernetes.io/cephfs"
I0831 02:51:00.203921    2113 plugins.go:291] Loaded volume plugin "kubernetes.io/downward-api"
I0831 02:51:00.204090    2113 plugins.go:291] Loaded volume plugin "kubernetes.io/fc"
I0831 02:51:00.204270    2113 plugins.go:291] Loaded volume plugin "kubernetes.io/flocker"
I0831 02:51:00.204471    2113 plugins.go:291] Loaded volume plugin "kubernetes.io/azure-file"
I0831 02:51:00.204659    2113 plugins.go:291] Loaded volume plugin "kubernetes.io/configmap"
I0831 02:51:00.214012    2113 server.go:632] Started kubelet
E0831 02:51:00.214829    2113 kubelet.go:956] Image garbage collection failed: unable to find data for container /
E0831 02:51:00.215007    2113 event.go:202] Unable to write event: 'Post http://localhost:8080/api/v1/namespaces/default/events: dial tcp 127.0.0.1:8080: connection refused' (may retry after sleeping)
I0831 02:51:00.215050    2113 server.go:109] Starting to listen on 0.0.0.0:10250
I0831 02:51:00.215455    2113 server.go:126] Starting to listen read-only on 0.0.0.0:10255
E0831 02:51:00.238862    2113 kubelet.go:1016] Failed to start ContainerManager system validation failed - Following Cgroup subsystem not mounted: [cpuset]
I0831 02:51:00.239134    2113 manager.go:123] Starting to sync pod status with apiserver
I0831 02:51:00.239250    2113 kubelet.go:2356] Starting kubelet main sync loop.
[... cut ...]
E0831 02:51:00.383532    2113 manager.go:212] Docker container factory registration failed: docker found, but not using native exec driver.
I0831 02:51:00.413614    2113 factory.go:97] Registering Raw factory
I0831 02:51:00.444413    2113 kubelet.go:2750] Recording NodeHasSufficientDisk event message for node 172.17.12.79
I0831 02:51:00.449813    2113 kubelet.go:1134] Attempting to register node 172.17.12.79
I0831 02:51:00.475042    2113 kubelet.go:1137] Unable to register 172.17.12.79 with the apiserver: Post http://localhost:8080/api/v1/nodes: dial tcp 127.0.0.1:8080: connection refused`

luxas, 2016-08-31:
> skipping pod synchronization - [Failed to start ContainerManager system validation failed - Following Cgroup subsystem not mounted: [cpuset] container runtime is down]

@pakeha-kiwi You have to set `cgroup_enable=cpuset` in `/boot/cmdline.txt`

erikthorselius, 2016-08-31:
How do i fix the "docker-flannel overlay was not installed by default" problem?

erikthorselius, 2016-08-31:
Looks like the dropin fails when removing the interface.

```bash
systemctl status docker
● docker.service - Docker Application Container Engine
   Loaded: loaded (/lib/systemd/system/docker.service; enabled)
  Drop-In: /usr/lib/systemd/system/docker.service.d
           └─docker-flannel.conf
        /etc/systemd/system/docker.service.d
           └─overlay.conf
   Active: active (running) since Wed 2016-08-31 23:47:48 EEST; 2min 33s ago
     Docs: https://docs.docker.com
  Process: 3091 ExecStartPre=/bin/sh -c ifconfig docker0 down; brctl delbr docker0 (code=exited, status=1/FAILURE)
```

luxas, 2016-08-31:
Is brctl installed?

erikthorselius, 2016-08-31:
yes
```bash
$ brctl
Usage: brctl [commands]
commands:
	addbr     	<bridge>		add bridge
	delbr     	<bridge>		delete bridge
	addif     	<bridge> <device>	add interface to bridge
	delif     	<bridge> <device>	delete interface from bridge
	hairpin   	<bridge> <port> {on|off}	turn hairpin on/off
	setageing 	<bridge> <time>		set ageing time
	setbridgeprio	<bridge> <prio>		set bridge priority
	setfd     	<bridge> <time>		set bridge forward delay
	sethello  	<bridge> <time>		set hello time
	setmaxage 	<bridge> <time>		set max message age
	setpathcost	<bridge> <port> <cost>	set path cost
	setportprio	<bridge> <port> <prio>	set port priority
	show      	[ <bridge> ]		show a list of bridges
	showmacs  	<bridge>		show a list of mac addrs
	showstp   	<bridge>		show bridge stp info
	stp       	<bridge> {on|off}	turn stp on/off
```

erikthorselius, 2016-08-31:
I'm new to kubernetes and flannel but what I can see from ifconfig it looks like the usal docker0 interface. 

```bash
ifconfig
docker0   Link encap:Ethernet  HWaddr 02:42:62:14:64:0f
          inet addr:172.17.0.1  Bcast:0.0.0.0  Mask:255.255.0.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
```
but the containers looks like they are live and kicking
```bash
docker -H unix:///var/run/system-docker.sock ps
CONTAINER ID        IMAGE                     COMMAND                  CREATED             STATUS              PORTS               NAMES
d33d6ae229b3        kubernetesonarm/etcd      "/usr/local/bin/etcd "   55 minutes ago      Up 55 minutes                           k8s-etcd
fd596c1f48f9        kubernetesonarm/flannel   "/flanneld --etcd-end"   55 minutes ago      Up 55 minutes                           k8s-flannel
```

pakeha-kiwi, 2016-09-01:
thanks @luxas, that got etcd/flannel going so my cluster works now

1. it seems I don't have the kubectl binary but that was mentioned in another thread I think
2. if you're using a remote kubectl, what are the creds to interrogate the Apimangler on https://<cluster-ip>
3. how do you figure out what port the Dashboard is NATed to?

luxas, 2016-09-01:
The kubectl binary is downloadable from with 
```
curl -sSL https://storage.googleapis.com/kubernetes-release/release/v1.2.0/bin/linux/arm/kubectl > /usr/local/bin/kubectl
```

To use remote kubectl, just set `KUBERNETES_MASTER=http://{rpi-ip}:8080` or use `-s http://{rpi-ip}:8080`

For dashboard, visit `http://{master-ip}:8080/api/v1/proxy/namespaces/kube-system/services/kubernetes-dashboard`

erikthorselius, 2016-09-01:
I solved my problem with flannel by changing the start of `docker-flannel.conf` to
```Bash
[Unit]
After=flannel.service
Requires=flannel.service
```

Don't know why it did not work before but now it works...

erikthorselius, 2016-09-01:
After doing it on the rest of the cluster I realized I was wrong. 

```Bash
sudo systemctl status docker
● docker.service - Docker Application Container Engine
   Loaded: loaded (/lib/systemd/system/docker.service; enabled)
  Drop-In: /usr/lib/systemd/system/docker.service.d
           └─docker-flannel.conf
        /etc/systemd/system/docker.service.d
           └─overlay.conf
   Active: active (running) since Wed 2016-08-31 17:39:56 EEST; 17h ago
     Docs: https://docs.docker.com
  Process: 455 ExecStartPre=/bin/sh -c ifconfig docker0 down; brctl delbr docker0 (code=exited, status=1/FAILURE)
 Main PID: 471 (dockerd)
....
HypriotOS/armv7: pirate@cluster-node02 in ~
$ sudo rm /etc/systemd/system/docker.service.d/overlay.conf
HypriotOS/armv7: pirate@cluster-node02 in ~
$ sudo systemctl daemon-reload
suHypriotOS/armv7: pirate@cluster-node02 in ~
$ sudo ifconfig docker0
docker0   Link encap:Ethernet  HWaddr 02:42:ff:18:2d:84
          inet addr:172.17.0.1  Bcast:0.0.0.0  Mask:255.255.0.0
          inet6 addr: fe80::42:ffff:fe18:2d84/64 Scope:Link
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:31 errors:0 dropped:0 overruns:0 frame:0
          TX packets:36 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:2865 (2.7 KiB)  TX bytes:5462 (5.3 KiB)

HypriotOS/armv7: pirate@cluster-node02 in ~
$ sudo systemctl restart docker
HypriotOS/armv7: pirate@cluster-node02 in ~
$ ifconfig docker0
docker0   Link encap:Ethernet  HWaddr 02:42:93:2b:eb:d7
          inet addr:10.1.86.1  Bcast:0.0.0.0  Mask:255.255.255.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
```

Now the network is handed over between docker and flannel.

pakeha-kiwi, 2016-09-01:
Geting closer but I seem to be having some kind of internal proxy issue when I call the master from my Mac:

`$ kubectl cluster-info
Kubernetes master is running at http://localhost:8080
Dashboard is running at http://localhost:8080/api/v1/proxy/namespaces/kube-system/services/kubernetes-dashboard,

`http://172.17.12.65:8080/api/v1/proxy/namespaces/kube-system/services/kubernetes-dashboard`

`Error: 'dial tcp 172.18.0.2:9090: no route to host'
Trying to reach: 'http://172.18.0.2:9090/'`

pakeha-kiwi, 2016-09-01:
hi @erikthorselius, is my issue the same the same as the one you fixed? I tried removing the overlay.conf file but that caused other issues and the docker0 IP remained the same. Were you saying that your docker-flannel.conf solution was wrong?

bialad, 2016-09-01:
Will the fixes from this issue be included in v0.8.0?

phixor, 2016-09-01:
aren't these ones specific to 1.0 Blackbeard?

luxas, 2016-09-02:
@bialad v0.8.0 will use my "official" code [docker-multinode](https://github.com/kubernetes/kube-deploy/tree/master/docker-multinode)

You may test that also if you want to

bialad, 2016-09-02:
@luxas I've actually been running my RPI cluster using docker-multinode so far, but since that repo doesn't use a release schedule I've had some issues with bugs since I'm always pulling the latest commit when booting my RPIs. I'd figure that I'd use this repo for setting up my core RPI cluster, and docker-multinode as a way to create amd64 worker nodes as vms in a windows server. Don't know if that's even possible, but time will tell. ;)

What do you mean by "my official" though? I've viewed this repo as a wrapper for kube-deploy, with stable releases. That's why I'm hopping for hypriot v1.0 stability to the v0.8.0 release.

anthillmob, 2016-09-25:
@luxas my k8s was working great until a reboot today. I've been having network connectivity issues and have just noticed that IPv4 has been disabled on some nodes. Specifying a master in IPv6 doesn't seem to work. Any ideas?

I see from an earlier post in June that Docker 1.5. supported the new protocol but we are running 1.12 whichis newer so am doubly confused ;-)

ebagdasa, 2016-10-09:
I didn't find brctl in .deb installation, should I install it manually?

MathiasRenner, 2016-10-14:
For me, after a reboot of a worker node, the routing is broken (all K8s containers, which have been running before the reboot, are up again -> fine).

Of the list in the first post of this thread, Flannel seems to be the only thing that is important for me (I don't mount anything, cgroup_enable=cpuset, docker pull works fine etc.). How do I get 2. and 3. implemented? ping @saturnism 

@erikthorselius Where resides the `docker-flannel.conf` you mention? Can't find it on `/etc/systemd/system/`, and there's no `flannel` container running.

Here some logs:
- The proxy container of the worker node: https://gist.github.com/MathiasRenner/f0ce5cf117e06c7e7033ef4b1da629e4
- The master node Log file: https://gist.github.com/MathiasRenner/8cf56a4f91419b764cdff9cde6a0237b
- The worker node log file: https://gist.github.com/MathiasRenner/a8b86b0685d2e8d2d4931e0d2e5c327b

saturnism, 2016-11-16:
Ah, I think I know where some of the confusions w/ flannels are coming from. I was working w/ kubernetes on arm 0.7. Flannel configuration was here:
https://github.com/luxas/kubernetes-on-arm/blob/release-0.7/sdcard/rootfs/kube-systemd/etc/kubernetes/dropins/docker-flannel.conf

I think it's changed since 0.8