luxas/kubernetes-on-arm #124: Autostart system-docker after reboot

erikthorselius, 2016-09-05:
On the new docker-multinode i can't find that the system-docker starts up after reboot. In the docker-bootstrap.sh I can't see any trace of trying to autostart the system-docker.

```Bash
# Start a docker bootstrap for running etcd and flannel
kube::bootstrap::bootstrap_daemon() {

  kube::log::status "Launching docker bootstrap..."

  docker daemon \
    -H ${BOOTSTRAP_DOCKER_SOCK} \
    -p /var/run/docker-bootstrap.pid \
    --iptables=false \
    --ip-masq=false \
    --bridge=none \
    --graph=/var/lib/docker-bootstrap \
    --exec-root=/var/run/docker-bootstrap \
      2> /var/log/docker-bootstrap.log \
      1> /dev/null &

  # Wait for docker bootstrap to start by "docker ps"-ing every second
  local SECONDS=0
  while [[ $(docker -H ${BOOTSTRAP_DOCKER_SOCK} ps 2>&1 1>/dev/null; echo $?) != 0 ]]; do
    ((SECONDS++))
    if [[ ${SECONDS} == ${TIMEOUT_FOR_SERVICES} ]]; then
      kube::log::fatal "docker bootstrap failed to start. Exiting..."
    fi
    sleep 1
  done
}
```

luxas, 2016-09-05:
Thanks for your will to improve this project! :)

I've clearly pointed that out in a few places that reboots won't work since no systemd script or something like that is installed.
https://github.com/luxas/kubernetes-on-arm#reboots is one of them.

For me, this is kind of a temporary release with the latest _stable_ stuff before I've got the really nice features I and others have been working on lately. You can see the roadmap for more details and look at sig-cluster-lifecycle at the kubernetes slack channel.

erikthorselius, 2016-09-05:
Aaa sorry my bad! Anyway I just added the old system-docker.socket and system-docker.service (renamed it to docker-bootstrap). It works really good.

closed 2016-09-05 by erikthorselius

erikthorselius, 2016-09-06:
I got I email how I fixed it. So I made a small gist with my changes. https://gist.github.com/erikthorselius/6724e82e40dbe6d2617378d77490bb07

luxas, 2016-09-06:
You could add a document about how to do it in `docs`...
Can you contribute that?

erikthorselius, 2016-09-06:
Sure, my only question if you want the script to be named system-docker (as before) or docker-bootstrap as it looks in master.sh?