luxas/kubernetes-on-arm #129: Dashboard addon doesn't exists [v0.8.0]

nlamirault, 2016-09-12:
Hi,
i've just created a K8S master. I try to enable the dashboard i've got this message : 
```
$ sudo kube-config enable-addon dashboard
This addon doesn't exist: dashboard
$ sudo kube-config info
Architecture: armv7l
Kernel: Linux 4.4.15
CPU: 4 cores x 1200 MHz

Used RAM Memory: 647 MiB
RAM Memory: 861 MiB

Used disk space: 1.6GB (1621200 KB)
Free disk space: 51GB (52926016 KB)

SD Card/deb package was built: 04-09-2016 14:39

kubernetes-on-arm: 
Latest commit: c64b05d
Version: 0.8.0

systemd version: v215
docker version: v1.12.1
kubernetes server version: v1.3.6

CPU Time (minutes):
kubelet: 9:39
kubelet has been up for: 2 hours
proxy: 0:25
apiserver: 1:59
controller-manager: 1:57
scheduler: 0:15

```

Any idea ? Thanks.

luxas, 2016-09-12:
It's enabled by default :)

nlamirault, 2016-09-12:
it seemed to me too.
But , I've got this error on the master : 
```
http://192.168.1.23:8080/api/v1/proxy/namespaces/kube-system/services/kubernetes-dashboard
{
  "kind": "Status",
  "apiVersion": "v1",
  "metadata": {},
  "status": "Failure",
  "message": "endpoints \"kubernetes-dashboard\" not found",
  "reason": "NotFound",
  "details": {
    "name": "kubernetes-dashboard",
    "kind": "endpoints"
  },
  "code": 404
}
```

nlamirault, 2016-09-14:
```
$ kubectl get services --all-namespaces 
NAMESPACE     NAME                   CLUSTER-IP   EXTERNAL-IP   PORT(S)         AGE
default       kubernetes             10.0.0.1     <none>        443/TCP         1d
kube-system   kube-dns               10.0.0.10    <none>        53/UDP,53/TCP   1d
kube-system   kubernetes-dashboard   10.0.0.110   <none>        80/TCP          1d
```

Dashboard is present. Sorry for noise.

closed 2016-09-14 by nlamirault

breuill, 2017-05-19:
Hi guys,

How did U solved this ?
I have started a 3 nodes cluster and unable to add dashboard.

Thank you for help :)

```
$ sudo kube-config info
Architecture: armv7l
Kernel: Linux 4.4.50
CPU: 4 cores x 1200 MHz

Used RAM Memory: 948 MiB
RAM Memory: 972 MiB

Used disk space: 2.5GB (2547688 KB)
Free disk space: 11GB (11033880 KB)

SD Card/deb package was built: 04-09-2016 14:39

kubernetes-on-arm: 
Latest commit: c64b05d
Version: 0.8.0

systemd version: v215
docker version: v17.03.0-ce
kubernetes server version: v1.3.6

CPU Time (minutes):
kubelet: 2:52
kubelet has been up for: 36 minutes
proxy: 0:07
apiserver: 0:48
controller-manager: 0:30
scheduler: 0:05

$ sudo kube-config enable-addon dashboard
This addon doesn't exist: dashboard

$ kubectl get --all-namespaces pods,rc,services
NAMESPACE     NAME                                 READY        STATUS             RESTARTS   AGE
kube-system   k8s-master-192.168.144.153           4/4          Running            44         36m
kube-system   k8s-proxy-192.168.144.153            1/1          Running            2          36m
kube-system   kube-addon-manager-192.168.144.153   1/2          ImagePullBackOff   2          31m
NAMESPACE     NAME                                 CLUSTER-IP   EXTERNAL-IP        PORT(S)    AGE
default       kubernetes                           10.0.0.1     <none>             443/TCP    36m

```

breuill, 2017-05-19:
More details with the describes of the failed pod:
```
$ kubectl describe pod "kube-addon-manager-192.168.144.153" --namespace=kube-system
Name:		kube-addon-manager-192.168.144.153
Namespace:	kube-system
Node:		192.168.144.153/192.168.144.153
Start Time:	Fri, 19 May 2017 06:37:25 +0200
Labels:		<none>
Status:		Pending
IP:		192.168.144.153
Controllers:	<none>
Containers:
  kube-addon-manager:
    Container ID:	
    Image:		gcr.io/google-containers/kube-addon-manager-arm:v4
    Image ID:		
    Port:		
    Requests:
      cpu:			5m
      memory:			50Mi
    State:			Waiting
      Reason:			ImagePullBackOff
    Ready:			False
    Restart Count:		0
    Environment Variables:	<none>
  kube-addon-manager-data:
    Container ID:	docker://81de175ffa03027bf965789d7876ca5d2ef406110d174747fb435c0e739d0333
    Image:		gcr.io/google_containers/hyperkube-arm:v1.3.6
    Image ID:		docker://sha256:0ccd589647b2c403d9aa64dbbf4374087631464934647559f5e137a0f0096d26
    Port:		
    Command:
      /copy-addons.sh
    State:			Running
      Started:			Fri, 19 May 2017 06:43:46 +0200
    Last State:			Terminated
      Reason:			Error
      Message:			Error on reading termination-log /var/lib/kubelet/pods/12985f6c2276246edebe61849d91e5be/containers/kube-addon-manager-data/49ae3b54: open /var/lib/kubelet/pods/12985f6c2276246edebe61849d91e5be/containers/kube-addon-manager-data/49ae3b54: no such file or directory
      Exit Code:		255
      Started:			Thu, 18 May 2017 22:57:15 +0200
      Finished:			Thu, 18 May 2017 23:12:52 +0200
    Ready:			True
    Restart Count:		2
    Environment Variables:	<none>
Conditions:
  Type		Status
  Initialized 	True 
  Ready 	False 
  PodScheduled 	True 
Volumes:
  addons:
    Type:	EmptyDir (a temporary directory that shares a pod's lifetime)
    Medium:	
QoS Tier:	Burstable
No events.
```

Thank's again :)

luxas, 2017-05-19:
Please use kubeadm instead on your Raspberry Pi, that's now the best way to do this.

This project is deprecated in favor for kubeadm which I'm now maintaining.

Cheers!

breuill, 2017-05-19:
Thank's for your answer; I'm using this one because I have some RPi 1 (arm v6) and the new one doesn't have a build for this ARM release isn't it ?

luxas, 2017-05-19:
Sorry, but I don't have time to look into using Kubernetes on a Pi 1. Worked up to v1.5, but we had to drop support unfortunately...

It's definitely possible to do if you spend some time hacking like I did though ;)

breuill, 2017-05-19:
Ok thanks, I understand, I will look this and if I find some hacks, I will share with you :)