luxas/kubernetes-on-arm #131: .deb package installation fails

mitchhh22, 2016-09-24:
I'm trying to install the latest on a rpi zero and everytime the script hangs here:
`HypriotOS/armv6: pirate@kube-node-2 in ~
$ sudo kube-config install
Which board is this running on? Options: [bananapro, cubietruck, generic, odroid-c2, parallella, pine64, rpi-2, rpi-3, rpi]. rpi
Which OS do you have? Options: [archlinux, debian, hypriotos]. hypriotos
Downloading kube-deploy...
Cloning into '/etc/kubernetes/kube-deploy'...
remote: Counting objects: 7818, done.
remote: Compressing objects: 100% (18/18), done.
remote: Total 7818 (delta 7), reused 0 (delta 0), pack-reused 7798
Receiving objects: 100% (7818/7818), 11.45 MiB | 170.00 KiB/s, done.
Resolving deltas: 100% (2616/2616), done.
Checking connectivity... done.
Checking out files: 100% (3503/3503), done.
Checking out files: 100% (9/9), done.
Note: checking out 'e4c1bccca6897e4df6221af6b60fbb80e5b6708f'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b new_branch_name

HEAD is now at e4c1bcc... Merge pull request #218 from luxas/stability`

ctrl-c and retrying hangs here:

`HypriotOS/armv6: pirate@kube-node-2 in ~
$ sudo kube-config install
rm: cannot remove ‘/etc/systemd/system/docker.service.d/overlay.conf’: No such file or directory
Downloading kube-deploy...
fatal: destination path '/etc/kubernetes/kube-deploy' already exists and is not an empty directory.`

Seems to happen on both hypriotos and raspbian for me.

mitchhh22, 2016-09-30:
I just installed the latest version of Raspbian and get hungup on the kube-config install:

```
pi@kubepi:~ $ screen -r
Which board is this running on? Options: [bananapro, cubietruck, generic, odroid-c2, parallella, pine64, rpi-2, rpi-3, rpi]. rpi-3
Which OS do you have? Options: [archlinux, debian, hypriotos]. debian
/etc/kubernetes/env/os/debian.sh: line 70: syntax error near unexpected token `}'
/etc/kubernetes/env/os/debian.sh: line 70: `}'
Downloading kube-deploy...
Cloning into '/etc/kubernetes/kube-deploy'...
remote: Counting objects: 7818, done.
remote: Total 7818 (delta 0), reused 0 (delta 0), pack-reused 7818
Receiving objects: 100% (7818/7818), 11.44 MiB | 895.00 KiB/s, done.
Resolving deltas: 100% (2620/2620), done.
Checking connectivity... done.
Note: checking out 'e4c1bccca6897e4df6221af6b60fbb80e5b6708f'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b new_branch_name

HEAD is now at e4c1bcc... Merge pull request #218 from luxas/stability
What hostname do you want? Defaults to kubepi.
Which timezone should be set? Defaults to Europe/Helsinki. America/Los_Angeles
Which storage driver do you want? Defaults to overlay.
```

any idea why?

codecounselor, 2016-11-08:
I had to install docker first for raspian then it works
https://docs.docker.com/engine/installation/linux/raspbian/