luxas/kubernetes-on-arm #134: HA not working, right?

MathiasRenner, 2016-09-30:
@luxas High Availability on ARM is not yet working, right? I saw this on your Roadmap.

I am having hard times starting a service in a high available manner. I tried replication controllers,  `kubectl run ... expose ...` etc. - without success.

Of course, I can expose a single service from a single node. But high availability requires any service being available on any node, even if I http-request a node that does not run the container/pod I am requesting. Docker Swarm Mode 1.12.1 has built in Mesh routing via Ingress and can do exactly this. 

Do you have any ideas how to achieve this in K8s?

luxas, 2016-10-01:
High Available for the Kubernetes components themselves aren't the same thing as high availability for the apps running on top of Kubernetes, right?

In the roadmap, I meant that right now the master is the SPOF, and multiple masters should be supported in future releases.

High availability for apps should be as easy as creating a RC with replicas>1, and exposing the RC as a service. Then the service will load-balance the requests to all endpoints.

Does this answer your question? Otherwise it's clearly a kubernetes/kubernetes issue I think

luxas, 2016-10-05:
We're working on HA in sig-cluster-lifecycle right now with kubeadm, but it's a highly complex task with a lot of corner cases and stuff, so we don't have a ETA yet.

Follow us on Slack: https://kubernetes.slack.com/messages/sig-cluster-lifecycle/ and/or attend our meetings if you're interested.

MathiasRenner, 2016-10-06:
All right, that's all info I was looking for to add to my research project ("HA performance of K8s on ARM"). Thank you!

closed 2016-10-06 by MathiasRenner

luxas, 2016-10-12:
Just for my curiosity... can you send the research paper to me when it's in beta version or done?
It would be nice to read what you've written about it...

(We can take this conversation by email as well)

MathiasRenner, 2016-10-13:
I send you an email :)

MathiasRenner, 2016-10-22:
@luxas Did you get my email from 13.10. ? I could send you the beta version of the paper, if you confirm the email.

luxas, 2016-10-23:
I'm sorry for not responding, first I waited quite a long time for the email to arrive... just to now find out that MS Outlook SmartScreen or something like that had marked it as spam :(.

And this week I've been on a journey without internet, so responding now. 

Yes, please send the paper. 
I'll be more careful now :)