luxas/kubernetes-on-arm #135: Unable to find image 'gcr.io/google_containers/etcd-arm:2.2.5' locally

Slahser, 2016-10-02:
i know there is a issue like this before. 

Description: 

In my country there is country-side-firewall..

i got `Unable to find image 'gcr.io/google_containers/etcd-arm:2.2.5' locally` when `kube-config enable-master`

What i did:

make a Registry & tag `gcr.io/google_containers/etcd-arm:2.2.5` locally

```
REPOSITORY                                     TAG                 IMAGE ID            CREATED             SIZE
myregistry.com/google_containers/etcd-arm   2.2.5               23e15ba74b83        5 months ago        24.91 MB
gcr.io/google_containers/etcd-arm              2.2.5               23e15ba74b83        5 months ago        24.91 MB
```

it appears also.

change dns ,not work.
set docker_opts proxy things,not work.
then i config shadowsocks on my pi cluster,it does not work.

any ways to help?

i am wondering why tag locally not work...

```
$ kube-config enable-master
+++ [1003 06:51:29] K8S_VERSION is set to: v1.3.6
+++ [1003 06:51:29] ETCD_VERSION is set to: 2.2.5
+++ [1003 06:51:29] FLANNEL_VERSION is set to: v0.6.1
+++ [1003 06:51:29] FLANNEL_IPMASQ is set to: true
+++ [1003 06:51:29] FLANNEL_NETWORK is set to: 10.1.0.0/16
+++ [1003 06:51:29] FLANNEL_BACKEND is set to: udp
+++ [1003 06:51:29] RESTART_POLICY is set to: unless-stopped
+++ [1003 06:51:29] MASTER_IP is set to: localhost
+++ [1003 06:51:29] ARCH is set to: arm
+++ [1003 06:51:29] IP_ADDRESS is set to: 192.168.31.222
+++ [1003 06:51:29] USE_CNI is set to: false
+++ [1003 06:51:29] USE_CONTAINERIZED is set to: false
+++ [1003 06:51:29] --------------------------------------------
+++ [1003 06:51:29] Killing all kubernetes containers...
+++ [1003 06:51:30] Launching docker bootstrap...
+++ [1003 06:51:32] Launching etcd...
Unable to find image 'gcr.io/google_containers/etcd-arm:2.2.5' locally
docker: Error response from daemon: Get https://gcr.io/v1/_ping: dial tcp 74.125.204.82:443: i/o timeout.
See 'docker run --help'.
!!! [1003 06:53:03] etcd failed to start. Exiting...
```

steve-wormley, 2016-10-04:
I saw this when it setup Kubernets DNS server as first in /etc/resolv.conf and was failing even though other resolvers were listed. I removed the configuration for that dns server(somewhere under /etc/, don't remember where) and rebooted. Once the K8s dns was up then I added the line back to that file and resolv.conf.

luxas, 2016-10-05:
Are you on HypriotOS, Raspbian or Arch Linux?

Slahser, 2016-10-05:
it is HypriotOS

closed 2017-05-11 by Slahser