luxas/kubernetes-on-arm #137: error restarting haproxy -- ./haproxy_reload: line 26: syntax error: unexpected redirection

steve-wormley, 2016-10-04:
error restarting haproxy -- ./haproxy_reload: line 26: syntax error: unexpected redirection

I'm seeing the above error when trying to setup the loadbalancer service, I'm not sure if this is the only problem but it's the easiest one to fix.

It appears to be coming because the image  kubernetesonarm/loadbalancer is based on busybox but the haproxy_reload script is using the <<< redirect which busybox does not support.

26: socat /tmp/haproxy - <<< "show servers state" > /var/state/haproxy/global

I'm not sure where the source of the kubernetesonarm/loadbalancer image is, so I'm unable to provide a pull request to fix it. but it should be as easy as 

echo "show servers state" | socat /tmp/haproxy - >/var/state/haproxy/global

luxas, 2016-10-05:
`haproxy_reload` source is here: https://github.com/luxas/kubernetes-on-arm/blob/master/addons/images/loadbalancer/haproxy_reload

Feel free to send a pull request!

nicolaiskogheim, 2016-12-26:
Just thought I'd reference the PR for this, here.

#138