luxas/kubernetes-on-arm #144: Cluster state taking a long time to update...

derailed, 2016-11-17:
Running ku 1.4.1 on Rpi with kubeadm-arm 1.5 on Hypriot 1.0.1. I've noticed that when I unplug the ethernet cable on one of the minions it takes a really long time for k8s to a) figure out the node is gone and b) pick up the given pods are no longer running and need to be instantiated on another available node. 

Does any one else see this or is something not configed right on my cluster?

For example once the ethernet jack is unpluged

kubectl get node -> Takes about ~2mins to pick a node not ready
kubectl get pod   -> Pod(s) show running status on the disabled node for over 5 mins and then the replication takes place.

Not sure why there is such a big lag in the heartbeat protocol? Wondering is something is not quiet configed right with etcd or flannel on my pi cluster??

luxas, 2016-11-17:
This is actually the default configuration for the controller-manager: http://kubernetes.io/docs/admin/kube-controller-manager/

```
--pod-eviction-timeout duration                                     The grace period for deleting pods on failed nodes. (default 5m0s)
--node-monitor-grace-period duration                                Amount of time which we allow running Node to be unresponsive before marking it unhealthy. Must be N times more than kubelet's nodeStatusUpdateFrequency, where N means number of retries allowed for kubelet to post node status. (default 40s)
```

This is more of a "feature" of k8s, to not be so aggressive in evicting pods from nodes, but for demos, raspberry pi clusters, etc. it is really helpful to set these to a lower value.
You can just edit `/etc/kubernetes/manifests/kube-controller-manager.json` and add those flags with the values you'd like.

Hope it helps

derailed, 2016-11-17:
Right on point, exactly what I was looking for. Thanks for the insight
Lucas!