luxas/kubernetes-on-arm #145: Accessing instance devices...

derailed, 2016-11-17:
I am aware that k8s does not support docker option --device. So I've resorted to use the security context privileged:true when defining a pod. This however does not yield the expected results on having the container access the instance devices.  

NOTE: both the api server and kubelet are running with allow-privileged: true

I can run the exact docker image on the node manually with --privileged: true and the container performs nominally. However launching the container via k8s, the application can no longer access the devices. Looking further, and running docker inspect on the docker and k8s container, I can see the k8s is running in privileged mode as expected however there is a delta in both the security context and devices settings. Docker says devices: [] while K8s container says devices: null. 
Thinking that might be the root cause of this issue but not super sure??

Has anyone been able to access /dev or /sys/class on the host instance from a k8s instantiated image? Any thoughts or workarounds?

luxas, 2016-11-17:
Have you tried using the hostPath volume?

derailed, 2016-11-17:
Tx Lucas! I did try the volume mounts and thought that would work, but
ended it being a dud.

I'll give it another rinse, perhaps I've missed something in the
excitement. I'll update with my findings.

derailed, 2016-11-18:
Again thank you Lucas for prompting me to revisit. You Rock!!

Turns out one on the mount points  symlinked to another dir that I was not
mounting ;-(
My stuff is running like a champ now.

Thank so much for your contributions and excellent input!