luxas/kubernetes-on-arm #149: Raspberry Pi Kubernetes

mastermindg, 2016-12-26:
I'm trying to setup Kubernetes for the first time on my Pi. I followed the README and installed kubeadm without any issues. When I attempt to setup the master I'm getting stuck at initialization. Here's the output from kubeadm:

```
$ kubeadm init --pod-network-cidr 10.244.0.0/16
[kubeadm] WARNING: kubeadm is in alpha, please do not use it for production clusters.
[preflight] Running pre-flight checks
[preflight] The system verification failed. Printing the output from the verification:
OS: Linux
KERNEL_VERSION: 4.4.39-hypriotos-v7+
CONFIG_NAMESPACES: enabled
CONFIG_NET_NS: enabled
CONFIG_PID_NS: enabled
CONFIG_IPC_NS: enabled
CONFIG_UTS_NS: enabled
CONFIG_CGROUPS: enabled
CONFIG_CGROUP_CPUACCT: enabled
CONFIG_CGROUP_DEVICE: enabled
CONFIG_CGROUP_FREEZER: enabled
CONFIG_CGROUP_SCHED: enabled
CONFIG_CPUSETS: enabled
CONFIG_MEMCG: enabled
CONFIG_INET: enabled
CONFIG_EXT4_FS: enabled
CONFIG_PROC_FS: enabled
CONFIG_NETFILTER_XT_TARGET_REDIRECT: enabled (as module)
CONFIG_NETFILTER_XT_MATCH_COMMENT: enabled (as module)
CONFIG_OVERLAY_FS: enabled (as module)
CONFIG_AUFS_FS: not set - Required for aufs.
CONFIG_BLK_DEV_DM: enabled (as module)
CGROUPS_CPU: enabled
CGROUPS_CPUACCT: enabled
CGROUPS_CPUSET: missing
CGROUPS_DEVICES: enabled
CGROUPS_FREEZER: enabled
CGROUPS_MEMORY: enabled
DOCKER_VERSION: 1.12.5
DOCKER_GRAPH_DRIVER: overlay
[preflight] Some fatal errors occurred:
	missing cgroups: cpuset
[preflight] If you know what you are doing, you can skip pre-flight checks with `--skip-preflight-checks`
```

I tried bypassing this with --skip-preflight-checks but I'm getting hung up at:

```[apiclient] Created API client, waiting for the control plane to become ready```

What do I need to do to get the master node up and running here?

nicolaiskogheim, 2016-12-26:
This is an error you'll have to fix:
```
[preflight] Some fatal errors occurred:
	missing cgroups: cpuset
```

You need to set `cgroup_enable=memory` and `cgroup_enable=cpuset` in `/boot/cmdline.txt`.

mastermindg, 2016-12-26:
That worked great, thanks! Happy Holidays!

closed 2016-12-26 by mastermindg