luxas/kubernetes-on-arm #152: DNS & flannel not starting

anthillmob, 2017-01-22:
hi there, great project however my cluster build seems fine master starts ok and nodes can be added and removed Ok but the DNS pod seems stuck and the flannel network step fails

I have a Pi3 master and a mix of 3s and 2Bs for the nodes

how to diagnose this further please?

`$ kubectl get pods --all-namespaces
NAMESPACE     NAME                              READY     STATUS              RESTARTS   AGE

default       kube-flannel-ds-5kd4q             0/2       CrashLoopBackOff    4          44s
kube-system   dummy-2501624643-tttz4            1/1       Running             0          1m
kube-system   etcd-1                            1/1       Running             1          31s
kube-system   kube-apiserver-1                  1/1       Running             1          52s
kube-system   kube-controller-manager-1         1/1       Running             0          32s
kube-system   kube-discovery-1659614412-jbcqk   1/1       Running             0          1m
kube-system   kube-dns-4211557627-frf5k         0/4       ContainerCreating   0          1m
kube-system   kube-proxy-28tb9                  1/1       Running             0          1m
kube-system   kube-scheduler-1                  1/1       Running             1          27s`

srinat999, 2017-01-25:
Hey, I'm facing this issue too. Did you manage to solve it?

srinat999, 2017-01-25:
OK. So I fixed this. The culprit is a new release v0.7 release of flannel thats causing issues in arm. You can go back to the old release and things should work. Use this command for creating your flannel network:

```
 curl -sSL https://raw.githubusercontent.com/coreos/flannel/e3a9dbb45ad5f1539b30621086270699696dccd3/Documentation/kube-flannel.yml | sed "s/amd64/arm/g" | kubectl create -f -
```
Make sure you delete the existing flannel pods by ```kubectl delete daemonset kube-flannel-ds```

anthillmob, 2017-01-25:
That certainly fixed the flannel issue thanks.  Now it's just DNS that is broken I think?

```HypriotOS/armv7: pirate@1 in ~
$ kubectl get nodes
NAME      STATUS    AGE
1         Ready     7m
2         Ready     2m
3         Ready     2m
4         Ready     2m
5         Ready     2m
6         Ready     2m
HypriotOS/armv7: pirate@1 in ~
$ kubectl get pods --all-namespaces
NAMESPACE     NAME                              READY     STATUS              RESTARTS   AGE
kube-system   dummy-2501624643-p9ppw            1/1       Running             0          7m
kube-system   etcd-1                            1/1       Running             1          6m
kube-system   kube-apiserver-1                  1/1       Running             1          6m
kube-system   kube-controller-manager-1         1/1       Running             0          6m
kube-system   kube-discovery-1659614412-1shp2   1/1       Running             0          7m
kube-system   kube-dns-4211557627-z4wq8         0/4       ContainerCreating   0          7m
kube-system   kube-flannel-ds-2rkxl             2/2       Running             0          2m
kube-system   kube-flannel-ds-5914j             2/2       Running             0          2m
kube-system   kube-flannel-ds-88mv0             2/2       Running             0          6m
kube-system   kube-flannel-ds-j8r21             2/2       Running             0          2m
kube-system   kube-flannel-ds-mxjzb             2/2       Running             0          2m
kube-system   kube-flannel-ds-wpwk9             2/2       Running             0          2m
kube-system   kube-proxy-1gpln                  1/1       Running             0          2m
kube-system   kube-proxy-2knlw                  1/1       Running             0          2m
kube-system   kube-proxy-j5x7f                  1/1       Running             0          2m
kube-system   kube-proxy-jnwdf                  1/1       Running             0          7m
kube-system   kube-proxy-khg7q                  1/1       Running             0          2m
kube-system   kube-proxy-km82x                  1/1       Running             0          2m
kube-system   kube-scheduler-1                  1/1       Running             1          6m```

closed 2017-01-25 by anthillmob